ARMmbed/mbed-os #15387: Unused Variable Warnings on Nuvoton Targers

matthewg276, 2023-03-08:
### Description of defect
Nuvoton target libraries declare a uint32 variable within a header file (m480_ccap.h and nuc472_cap.h). It does not appear this variable is ever used and because of this, a large number of warnings are generating when compiling these targets since this file is included in M480.h. 

Example warning
```
[Warning] m480_ccap.h@86,17: 'u32EscapeFrame' defined but not used [-Wunused-variable]
```

#### Target(s) affected by this defect ?
- NUMAKER_PFM_M487
- NUMAKER_IOT_M487

Potentially NUC472 though I have not verified. 

#### Toolchain(s) (name and version) displaying this defect ?
- Using latest docker image ( ghcr.io/armmbed/mbed-os-env:master-2022.05.21t04.23.55) 
- using GCC ARM compiler

#### What version of Mbed-os are you using (tag or sha) ?
- mbed version 6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
- mbed cli 1 version included in docker container.

#### How is this defect reproduced ? 
- compile blinky program for NUMAKER_IOT_M487

0xc0170, 2023-03-08:
cc @ARMmbed/team-nuvoton

cyliangtw, 2023-03-09:
@matthewg276 , thank you for raising this issue.
NUC472 not include any capture header file, so NUC472 is fine.
About M487, this warning is fixed in PR #15388 .

matthewg276, 2023-03-17:
Confirming this has been fixed in latest master branch. Thank you!

closed 2023-03-17 by matthewg276