ARMmbed/mbed-os #15390: STM32F7 : CAN RX interrupt not triggering Mbed-OS 6.17.0

htex0189, 2023-03-13:
### Description of defect

#### Target(s) affected by this defect ?
STM32F7

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-studio
mbed-tools 7.59.0

#### How is this defect reproduced ?

mbedmain, 2023-03-13:
@htex0189 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What toolchain(s) are you using?
What Mbed OS version are you using?
How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

htex0189, 2023-03-13:
Solution
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM#readme

closed 2023-03-13 by htex0189