ARMmbed/mbed-os #15391: RAK3172 ADC pins don't work when deep sleep is enabled

goshawk22, 2023-03-13:
### Description of defect

#### Target(s) affected by this defect ?
RAK3172 (STM32WL)

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed studio 1.4.4

#### How is this defect reproduced ?

mbedmain, 2023-03-13:
@goshawk22 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What toolchain(s) are you using?
What Mbed OS version are you using?
How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

goshawk22, 2023-03-13:
The apparent hang after the first reading seems to have been caused by #15331. After fixing this, when deep sleep is enabled, analogin reports near zero values around 0.09.

jeromecoutant, 2023-03-13:
From STM32WL ref manual, seems that ADC configuration is not retained with Stop2 (deep sleep) power mode.
So idea could be:
- to disable deep sleep when ADC is configured
- to enable ADC only when a measurement is needed
- re-implement in your application hal_deepsleep(void)
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/sleep.c#L166
in order to save ADC registers before going in stop2 and restore them after

goshawk22, 2023-03-13:
With regards to point 1 and 2, if I put `AnalogIn voltage(PB_3)` inside a function so that it is called each time I want to read the adc pin, I get wrong values. It returns roughly 2.7 instead of the expected 3.3.

hallard, 2023-04-05:
@goshawk22 interesting answer from @jeromecoutant to rewrite hal_deepsleep for this one (and also for the serial) but I'm far from ST HAL guru, so if anyone have an example on how to reset 115200 on serial console after deep sleep, I take it.
Anyway I don't have the ADC issue mentioned here.

What I'm doing after deep sleep is to call `read_sensors()`  this as follow 

```cpp
bool read_sensors() 
{
  vdda = get_vdda();
  vbat = get_vbat();
  info("Vdda %dmV  Vbat %dmV  ", vdda, vbat);
}
```

and functions used are 

```cpp

// Defined globally
DigitalOut baten(BAT_EN, 0);
AnalogIn adc_vbat(BAT_MON);
uint16_t vdda = 0; // VDD Average in mV (calculated by internal ADC)
uint16_t vbat = 0; // Vbat in mV calculated with Resistor Divider

uint16_t get_vdd() 
{
  uint16_t vref_cal = *((const uint16_t*)VREFINT_CAL_ADDR);
  return ((float)VREFINT_CAL_VREF * vref_cal / (AnalogIn(ADC_VREF).read() * 4095.0));
}

uint16_t get_vdda() 
{
  uint32_t sum = 0;
  for (int i=0; i<8 ; i++) {
    sum += get_vdd();
  }
  vdda = sum >> 3 ;
  return vdda;
}

uint16_t get_vbat()
{
  float sum = 0;
  baten = 1;
  adc_vbat.set_reference_voltage((float) vdda / 1000.0f);
  ThisThread::sleep_for(5ms);
  for (int i=0; i<8 ; i++) {
    sum += adc_vbat.read_voltage() * 1.27 * 1000.0;
  }
  vbat = (uint16_t) (sum / 8.0f) ;
  baten = 0;
  return vbat;
}
```