ARMmbed/mbed-os #15393: Serial Communication Failure

AndrewJKnowles, 2023-03-16:
Hi there,

I am trying to use my laptop to communicate serially to an MBED NXP LPC1768 however, when I open my terminal emulator (CoolTerm) it informs me that no serial ports are available. The device its self seems to present its self as a mass storage device and I am able to access it through file explorer.

In an attempt to update the serial driver I have tried to download the serial driver given on this page... https://os.mbed.com/docs/mbed-os/v6.16/program-setup/windows-serial-driver.html however the download does not commence thus I cannot download the installer. I am aware that this installer is for windows 7 while newer version of window seem to run the driver “out of the box” however this does not seem to be the case (I am currently using Windows 10).

Could you please advice on how to over come this issue?

Also the email: mbed-support@arm.com does not allow emails to be sent to it unless the email address your sending from is on an approved list which seems counter productive.

AndrewJKnowles, 2023-03-16:
Target: MBED LPC1768
Target Firmware Version: rev141212
Toolchain: n/a
MBED OS: 6.17
MBED Studio version: 1.4.4
OS: Windows 10

Device Driver version: 
6.1.7600.16385 dated 06/2006 with the driver provider as Microsoft.

Terminal emulator: CoolTerm (Intel 64Bit)

How to replicate issue:
Connect device to pc/laptop. Device appears under “Portable Devices” rather than “Ports (COM & LPT)” in Device Manager. Device will appear in file explorer.

Open CoolTerm, CoolTerm indicates no useable comports detected.

Within MBED Studio, open serial monitor. No information is sent/received via serial.

0xc0170, 2023-03-20:
Hello @AndrewJKnowles,

the best would be to create a thread on mbed forum.

What firmware is running on your board? Would you try https://forums.mbed.com/t/new-beta-firmware-for-lpc1768/18917 ? But first mark  somewhere you current one (you can revert back in case something goes wrong).

You can review older firmwares: https://os.mbed.com/handbook/Firmware-LPC1768-LPC11U24

0xc0170, 2023-03-21:
The windows driver should be just removed , it just brings confusion for windows users. Often people just search for the driver first instead of the firmware version and its updates.

AndrewJKnowles, 2023-03-21:
Thank you for your help. I shall share my finding with the forum above.

Andrew

closed 2023-03-21 by AndrewJKnowles