ARMmbed/mbed-os #15397: Add control of special case FF_MAX_SS == FF_MIN_SS in FatFileSystem

bulanek, 2023-03-31:
### Description of defect

#### Target(s) affected by this defect ?
FatFileSystem

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

bulanek, 2023-03-31:
Please add control to
 int FATFileSystem::statvfs(const char *path, struct statvfs *buf):

#if FF_MAX_SS == FF_MIN_SS
    buf->f_bsize = FF_MAX_SS;
    buf->f_frsize = FF_MAX_SS;
#else
    buf->f_bsize = fs->ssize;
    buf->f_frsize =fs->ssize;

0xc0170, 2023-04-03:
Hi @bulanek , would you create a merge request to address this?