ARMmbed/mbed-os #15409: OSPI support for STM32H7 targets

wdx04, 2023-04-27:
Currently OSPI is only supported by a few STM32L4+, STM32L5 and STM32U5 targets.
Support for STM32H7 targets, such as STM32H7Ax/Bx/2x/3x are missing.

Only a few changes are needed to add OSPI support to STM32H7 targets.

1.In mbed-os/targets/TARGET_STM/ospi_api.c and mbed-os/targets/TARGET_STM/qspi_api.c
find the following line:
__HAL_RCC_OSPIM_CLK_ENABLE();
and change it to:
#if defined(TARGET_STM32H7)
    __HAL_RCC_OCTOSPIM_CLK_ENABLE();
#else
    __HAL_RCC_OSPIM_CLK_ENABLE();
#endif

2.in mbed-os/targets/TARGET_STM/TARGET_STM32H7/objects.h
modify the definition of struct qspi_s and add the struct ospi_s:
#if DEVICE_QSPI
struct qspi_s {
#if defined(OCTOSPI1)
    OSPI_HandleTypeDef handle;
#else
    QSPI_HandleTypeDef handle;
#endif
    QSPIName qspi;
    PinName io0;
    PinName io1;
    PinName io2;
    PinName io3;
    PinName sclk;
    PinName ssel;
};
#endif

#if DEVICE_OSPI
struct ospi_s {
    OSPI_HandleTypeDef handle;
    OSPIName ospi;
    PinName io0;
    PinName io1;
    PinName io2;
    PinName io3;
    PinName io4;
    PinName io5;
    PinName io6;
    PinName io7;
    PinName sclk;
    PinName ssel;
    PinName dqs;
};
#endif

I have tested the code with a custom STM32H7B0VB board which has a W25Q64 QSPI flash.
Please consider applying the changes to the next release.

Thanks.

mbedmain, 2023-04-27:
@wdx04 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2023-04-28:
Hi

> I have tested the code with a custom STM32H7B0VB board which has a W25Q64 QSPI flash.

Please, push the proposition in a patch, we will review it with pleasure :-)

Thx

PR #15414 (2023-05-08): Add OSPI support for STM32H7
### Summary of changes 

Enable OSPI for STM32H7 targets.

Added the necessary ospi_s struct to enable OSPI for STM32H7 targets
Modified qspi_s struct to enable OSPI in QSPI mode for STM32H7 targets
Use __HAL_RCC_OCTOSPIM_CLK_ENABLE() instead of __HAL_RCC_OSPIM_CLK_ENABLE() for STM32H7 targets. They are the same  thing, just with different names in STM32H7 and STM32L5/U5.

Fix #15409

#### Impact of changes

#### Migration actions required

### Documentation

None.

---------------------------------------------------------------------------------------------------- [... cut]