ARMmbed/mbed-os #15411: Can't compile to the "NUCLEO-H563ZI" board.

dojyorin, 2023-05-01:
### Description of defect
I can't compile to the "NUCLEO-H563ZI" board.

This board failed to compile even though the description on the [official website](https://os.mbed.com/platforms/st-nucleo-h563zi) says "supported by Mbed-OS 6.9".
I tried Mbed-OS v6.9, 6.15, and the latest 6.17, but they all failed.

#### Target(s) affected by this defect ?
- NUCLEO-H563ZI ( https://www.st.com/en/evaluation-tools/nucleo-h563zi.html )

#### Toolchain(s) (name and version) displaying this defect ?
- Keil Studio Cloud v1.6.11 ( https://studio.keil.arm.com )

#### What version of Mbed-os are you using (tag or sha) ?
- v6.9.0 ( https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.9.0 )
- v6.15.0 ( https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.15.0 )
- v6.17.0 ( https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.17.0 )

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
- Keil Studio Cloud v1.6.11 ( https://studio.keil.arm.com )

#### How is this defect reproduced ?
It occurs when "NUCLEO-H563ZI" is selected as the target and the Compile button is pressed.

Output Log in Keil Studio Cloud:
```log
Build started
Target "NUCLEO_H563ZI" is not recognized
Internal error.
Build failed
Build failed
```

jeromecoutant, 2023-05-02:
Hi
I am sorry but ST doesn't plan to support STM32H5 with mbed-os.
Regards,

dojyorin, 2023-05-03:
Thank you for reply.
I see... it's disappointing.
By the way, is it due to technical problems?

wdx04, 2023-05-04:
> Hi I am sorry but ST doesn't plan to support STM32H5 with mbed-os. Regards,

Does it mean that the STM32U5 will be the last series of STM32 to support mbed os?

jeromecoutant, 2023-05-05:
> By the way, is it due to technical problems?

no... Like ARM, ST has slowed down Mbed work.
https://forums.mbed.com/t/will-mbed-os-get-back-its-previous-pace-of-development/15839

> Does it mean that the STM32U5 will be the last series of STM32 to support mbed os?

I would say yes

dojyorin, 2023-05-06:
I understand the current situation with Mbed.
I would also consider using the STM32U5.
Thank you for kindly telling me.

closed 2023-05-06 by dojyorin