ARMmbed/mbed-os #15417: mbed studio: m2354: build failed

lunchboxwu, 2023-05-05:
### Description of defect

On ARM Studio, targeting NUMAKER-M2354 v1.1 (NU_M2354), mbed-os-example-blinky (
mbed-os 6.17.0) build failed with error log listed below.

#### Target(s) affected by this defect ?

MCU: Nuvoton M2354
Board: NUMAKER-M2354 v1.1 (NU_M2354)

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 6.17.0
Commit: 17dc3dc2e6e2817a8bd3df62f38583319f0e4fed

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio: 1.4.5

#### How is this defect reproduced ? 

1. Log in Mbed Studio
2. Click File > New Program > mbed-os-example-blinky > Add Program
3. Update library to mbed-os-6.17.0 if not yet
4. Specify NUMAKER-M2354 as target hardware
5. Build failed with log:
`Image: BUILD/NU_M2354/ARMC6\mbed-os-example-blinky.bin
0002874:CRITICAL:__main__:cannot read register r0 because core #0 is not halted
Traceback (most recent call last):
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 402, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\__main__.py", line 579, in do_flash
    file_format=self._args.format)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\file_programmer.py", line 157, in program
    self._loader.commit()
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\builder.py", line 464, in program
    flash_operation = self._chip_erase_program_double_buffer(progress_cb)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\builder.py", line 706, in _chip_erase_program_double_buffer
    self.flash.erase_all()
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\flash.py", line 345, in erase_all
    result = self._call_function_and_wait(self.flash_algo['pc_eraseAll'])
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\flash.py", line 604, in _call_function_and_wait
    return self.wait_for_completion()
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\flash\flash.py", line 600, in wait_for_completion
    return self.target.read_core_register('r0')
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\core\soc_target.py", line 173, in read_core_register
    return self.selected_core.read_core_register(id)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\cortex_m.py", line 911, in read_core_register
    regValue = self.read_core_register_raw(reg_info.index)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\cortex_m.py", line 928, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\cortex_m.py", line 949, in read_core_registers_raw
    return self._base_read_core_registers_raw(reg_list)
  File "c:\Users\lunch\AppData\Local\Mbed Studio\mbed-studio-tools\python\lib\site-packages\pyocd\coresight\cortex_m.py", line 967, in _base_read_core_registers_raw
    self.core_number))
pyocd.core.exceptions.CoreRegisterAccessError: cannot read register r0 because core #0 is not halted`

mbedmain, 2023-05-05:
@lunchboxwu thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2023-05-08:
cc @ARMmbed/team-nuvoton

ccli8, 2023-05-08:
The error log is related to pyocd. Currently, M2354 board's NuLink2 doesn't support flash or debug on Mbed Studio. Just build code only, that is, on Mbed Sutdio, click **Build program**, neither **Run program** nor **Debug program**. To flash code, go USB drag-and-drop programming.

cyliangtw, 2023-05-08:
@ccli8 
In fact, M2354 support pyocd in Jun 2021 at:
https://github.com/pyocd/pyOCD/blob/main/pyocd/target/builtin/target_M2354.py

please esnure Nu-Link2 with correct F/W and mbed-studio with up-to-date pyOCD pack.

cyliangtw, 2023-05-09:
After checking, Nu-Link2 default F/W is for normal drag-drop & mbed gree-tea test support.
For M2354 mbed-studio debug support, you must update as [NuLink2_DAPLink_for_numaker_m2354.bin](https://github.com/OpenNuvoton/Nuvoton_Tools/blob/master/Latest_NuLink_Firmware/NuLink2_DAPLink_for_numaker_m2354.bin).
How to update Nu-Link2 F/W, please refer to the steps of sub topic `DAP Link Interface Firmware Update` at: https://os.mbed.com/platforms/NUMAKER-M2354/

ccli8, 2023-05-09:
> For M2354 mbed-studio debug support, you must update as [NuLink2_DAPLink_for_numaker_m2354.bin](https://github.com/OpenNuvoton/Nuvoton_Tools/blob/master/Latest_NuLink_Firmware/NuLink2_DAPLink_for_numaker_m2354.bin).

This is necessary. However, Mbed Studio's inbuilt pyOCD version **0.30.2** hasn't supported M2354. Upgrade this inbuilt pyOCD to latest version manually may break Mbed Studio. So Flash & Debug for M2354 are still not supported on Mbed Studio.

cyliangtw, 2023-05-09:
Manual upgrade pyocd is relative to python site package, for ex "`Python38\Lib\site-packages\pyocd`". My environment is pyocd v0.34.0 and get:
```
[cygwin~zephyr]$ pyocd list --targets | grep Nuvoton
  m2354kjfae                Nuvoton                 M2354KJFAE                              builtin 
  m252kg6ae                 Nuvoton                 M252KG6AE                               builtin 
  m263kiaae                 Nuvoton                 M263KIAAE                               builtin 
  m467hjhae                 Nuvoton                 M467HJHAE                               builtin 
  m487jidae                 Nuvoton                 M487JIDAE                               builtin 

```
In mbed studio environment console, it should reach python path and could manual update by "`pip install pyocd`" ?

ccli8, 2023-05-10:
> In mbed studio environment console, it should reach python path and could manual update by "pip install pyocd" ?

Yes. In Mbed Studio, open builtin console through **Terminal** > **New Terminal** and run like `pip install -U pyocd` to upgrade pyocd. But I don’t suggest doing it because as reported it will break Mbed Studio on pyocd relative files.

ccli8, 2023-05-31:
To enable flashing and debugging M2354 on Mbed Studio, please follow the flow:
1. Apply #15427 to fix Mbed OS application program doesn't start on the sector boundary. This is required by pyOCD Debug.

1. Follow [the procedure](https://os.mbed.com/platforms/NUMAKER-M2354/#dap-link-interface-firmware-update) to upgrade NuLink2 F/W to v7509 or later. After upgrade is finished and USB is reconnected, open  **NU_CFG.TXT** in the USB drive and check its content:
    - Version >= 7509 (`Version=7509`)
    - CMSIS DAP enabled (`CMSIS-DAP=1`).

    If either doesn't meet, apply one clean upgrade through ICP Tool instead of drag-and-drop upgrade as mentioned in *As While DAP Button Not Work* section in above procedure.

    **NOTE**: As of this writing, v7509 is not released yet.

1. Mbed Studio 1.4.5 + pyOCD 0.30.2 doesn't support M2354 natively. Go extended support way through CMSIS Pack (detailed below).

ccli8, 2023-05-31:
To extend pyOCD support for M2354 via CMSIS Pack, follow the procedure (verified on Windows 10):
1. Download [Nuvoton.NuMicro_DFP.1.3.20.pack](https://github.com/OpenNuvoton/cmsis-packs/tree/master/Nuvoton_DFP). Assume it is saved in **C:\Users\MY_ACCOUNT\Downloads**.

1. Cnnect M2354. Launch Mbed Studio.

1. Follow [Manage Custom Targets](https://os.mbed.com/docs/mbed-studio/current/targets/custom-targets.html). In **DebugFlags** field, override its content with `--target M2354KJFAE`. Click **Save All**.

1. Find **custom_targets.json** through **Help** > **Report an issue**. Click **Access Log** to open the directory containing **custom_targets.json**.

1. Exit Mbed Studio.

1. Open **custom_targets.json**. All content may place in single line. Try google **json format** to find one tool to re-arrange it.

1. Based on existent content, edit with the content. Save it.
    ```json
    {
        "13120000002B005104448580000009FA": {
            "debugFlags": "--target M2354KJFAE --pack \"C:\\Users\\MY_ACCOUNT\\Downloads\\Nuvoton.NuMicro_DFP.1.3.20.pack\"",
            "productCode": "1312",
            "debugTarget": "M2354KJFAE",
            "cmsisPack": "C:\\Users\\MY_ACCOUNT\\Downloads\\Nuvoton.NuMicro_DFP.1.3.20.pack"
        }
    }
    ```

1. Re-launch Mbed Studio. If everything goes fine, flashing and debugging M2354 should be OK.

cyliangtw, 2023-09-05:
This issue was fixed and could be closed.