ARMmbed/mbed-os #15418: STM32U5 firmware package need update

wdx04, 2023-05-08:
### Description of defect

I created a BurstSPI class for transfering image data from MCUs to TFT-LCDs via DMA. The class works fine with many STM32 MCUs including F1,F4,F7,L4,L5 and H7. But when using it with the STM32U5 MCU, the application will crash after the first successful DMA transfer, and the crash is caused by a bug in the HAL_DMA_IRQHandler function.

As of version 6.17, Mbed-os still ships with the initial 1.0.0 version of STM32U5 firmware package, which seems to be not very stable. I replaced it with the latest 1.2.0 version and recompile, the application now works as expected.

I think it's time to upgrade the bundled STM32U5 firmware package. It will also help us to port Mbed-os to newer STM32U5 products, such as NUCLEO-U5A5ZJ-Q.

Thanks.

#### Target(s) affected by this defect ?

NUCLEO-U575ZI-Q

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6.16

#### What version of Mbed-os are you using (tag or sha) ?

Mbed-os v6.17
SHA: 17dc3dc2e6e2817a8bd3df62f38583319f0e4fed

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.4.5

#### How is this defect reproduced ? 

1. Get the BurstSPI class from https://github.com/wdx04/BurstSPI
2. Create a minimal Mbed applicaiton
3. Create a BurstSPI instance and call fastWrite with some data

0xc0170, 2023-05-08:
cc @ARMmbed/team-st-mcd

multiplemonomials, 2023-05-08:
Woah, this library looks awesome!  I will have to see about integrating functionality like this into Mbed CE!

0xc0170, 2023-06-13:
> Could you try https://github.com/ARMmbed/mbed-os/pull/15420 ?

I'll close this as resolved.

closed 2023-06-13 by 0xc0170