ARMmbed/mbed-os #15428: Running `mbed-tools detect` on mac_os fails when a USB peripheral is missing some data

tommaso-zandrini, 2023-05-29:
### Description of defect

Running `mbed-tools detect` on mac_os fails when a USB peripheral (i.e. my USB mouse) does not provide some data (in my case, there is no `vendor_id` in the USB device properties)

in file `mbed_tools/devices/_internal/darwin/device_detector.py`
in function `_assemble_candidate_data`
the line `device_data.get("vendor_id", "")` produces a crash when the vendor_id is not provided by the peripheral 

(I expect the same possible issue with the next two lines, using the same method to get "product_id" and "serial_num")

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 7.59.0

#### How is this defect reproduced ? 

running `mbed-tools detect` with the device without vendor_id produces the crash
running `mbed-tools detect` after disconnecting that device works as expected

mbedmain, 2023-05-29:
@tommaso-zandrini thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

tommaso-zandrini, 2023-05-29:
close here, move to mbed-tools repo

closed 2023-06-13 by 0xc0170