ARMmbed/mbed-os #15434: invalid value '' in '-std=' in keil exported from mbed os

appalto-alexa, 2023-07-04:
compiling USBDevice.cpp...
error: invalid value '' in '-std='
note: use 'c++98' or 'c++03' for 'ISO C++ 1998 with amendments' standard
note: use 'gnu++98' or 'gnu++03' for 'ISO C++ 1998 with amendments and GNU extensions' standard
note: use 'c++11' for 'ISO C++ 2011 with amendments' standard
note: use 'gnu++11' for 'ISO C++ 2011 with amendments and GNU extensions' standard
note: use 'c++14' for 'ISO C++ 2014 with amendments' standard
note: use 'gnu++14' for 'ISO C++ 2014 with amendments and GNU extensions' standard
note: use 'c++17' for 'ISO C++ 2017 with amendments' standard
note: use 'gnu++17' for 'ISO C++ 2017 with amendments and GNU extensions' standard
note: use 'c++2a' for 'Working draft for ISO C++ 2020' standard
note: use 'gnu++2a' for 'Working draft for ISO C++ 2020 with GNU extensions' standard
compiling USBEndpoints_Nuvoton.cpp...

mbedmain, 2023-07-04:
@appalto-alexa thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

appalto-alexa, 2023-07-04:
Hi,
I am using NUVOTON M487 IOT board.
I exported keil code from mbed studio "Mbed Studio: 1.4.5"
uv version is  5.26.2 and toolchain is MDK-Lite v5.26.2.
When i compiled exported code, it gives 247 errors i.e.  error: invalid value '' in '-std='

closed 2023-07-05 by appalto-alexa