ARMmbed/mbed-os #15439: STM32H743 cannot use all the available RAM

pmancele, 2023-07-27:
### Description of defect

Hello,
The STM32H743 is supposed to have 1 MB of RAM, but it is separated in different sections that does not seems to be continuous.
When compiling and running an Mbed OS application we only have access to 512k of RAM.

The sections are defined like this in the linker script: 

 ```
MEMORY
{ 
  FLASH (rx)     : ORIGIN = MBED_APP_START, LENGTH = MBED_APP_SIZE
  DTCMRAM (rwx)  : ORIGIN = NVIC_RAM_VECTOR_ADDRESS + VECTORS_SIZE, LENGTH = 128K - VECTORS_SIZE
  RAM (xrw)      : ORIGIN = MBED_RAM_START, LENGTH = MBED_RAM_SIZE
  RAM_D2 (xrw)   : ORIGIN = 0x30000000, LENGTH = 288K
  RAM_D3 (xrw)   : ORIGIN = 0x38000000, LENGTH = 64K
  ITCMRAM (xrw)  : ORIGIN = 0x00000000, LENGTH = 64K
}
```
whith:
```
#define MBED_RAM_START  0x24000000
#define MBED_RAM_SIZE  0x80000  // 512 KB
```

**Do you know a way to have access to other area than the RAM, like the RAM_D2 and RAM_D3 ?** We need more than 512k for our project.
Maybe we can specify a RAM area when instantiating a new variable ?

#### Target(s) affected by this defect ?
NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.5

#### How is this defect reproduced ? 

Trying to use more than 512k of RAM on STM32h7x3 targets

Thank you !

pmancele, 2023-07-27:
Maybe @jeromecoutant know how to deals with it and use the other RAM sections ? :)

0xc0170, 2023-07-28:
cc @ARMmbed/team-st-mcd

0xc0170, 2023-07-28:
I remember NXP K64F has 2 RAM banks as well, RAM is split into two sections (if I am not wrong, you can't write through the boundaries). 

See the linker script: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld - you can review data and data 2 sections. 
The first lower RAM is used for interrupts, heap and the second bank (larger) for user/app data.

pmancele, 2023-07-31:
Thank you.

I managed to use the RAM_D2 by creating a section like that in the linker script : 

```
SECTIONS
{
    ...
    .map_counters (NOLOAD):
    {
        KEEP(*(.map_counters))
    } >RAM_D2
}
```

Then, in my code i can instantiate vairables at this area like this:

```C++
static uint32_t map_counter[110][110] __attribute__((section(".map_counters")));
```

JohnK1987, 2023-08-03:
Hello,

sorry, but information in your main post are wrong.
> MEMORY
> { 
> FLASH (rx)     : ORIGIN = MBED_APP_START, LENGTH = MBED_APP_SIZE
> DTCMRAM (rwx)  : ORIGIN = NVIC_RAM_VECTOR_ADDRESS + VECTORS_SIZE, LENGTH = 128K - VECTORS_SIZE
> RAM (xrw)      : ORIGIN = MBED_RAM_START, LENGTH = MBED_RAM_SIZE
> RAM_D2 (xrw)   : ORIGIN = 0x30000000, LENGTH = 288K
> RAM_D3 (xrw)   : ORIGIN = 0x38000000, LENGTH = 64K
> ITCMRAM (xrw)  : ORIGIN = 0x00000000, LENGTH = 64K
> }
> #### What version of Mbed-os are you using (tag or sha) ?
> mbed-os-6.17.0

You wrote about wrong memory settings in MbedOS 6.17, but this memory setting is definitely not from Mbed OS 6.17.
* [STM32H743xI.ld file](https://github.com/ARMmbed/mbed-os/blob/13f43cce521a2b07250b8cb3865367f6e47c1019/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/TOOLCHAIN_GCC_ARM/STM32H743xI.ld#L40-L48)
* [msis_nvic.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/cmsis_nvic.h)

The memory settings from your post is from Mbed OS 6.15 or lower because it Mbed OS 6.15.1 was done a correction - https://github.com/ARMmbed/mbed-os/pull/15130

BR, Jan

pmancele, 2023-08-04:
Hello @JohnK1987 

Indeed, to be exact we are using a custom target based on the STM32H753ZI. We are using Mbed OS 6.17.0 but the linker script of our custom targets comes from an older version of Mbed OS.

I didn't see that the linker script was updated, but the issue remains : it only use the 512k bank for user RAM.

pmancele, 2023-08-04:
I'm not sure if there are any other solutions besides the one I detailed previously. So, unless you have another idea, we can close the issue :)

jeromecoutant, 2023-08-08:
> I didn't see that the linker script was updated, but the issue remains : it only use the 512k bank for user RAM

Not fully true => Ethernet is using SRAM3: 
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/emac/TARGET_STM/stm32xx_emac.cpp#L113
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/TOOLCHAIN_GCC_ARM/STM32H743xI.ld#L235