ARMmbed/mbed-os #15441: Support for Nucleo H753ZI (again)

mbuchwald-atg, 2023-08-04:
### Description of defect

According to https://github.com/ARMmbed/mbed-os/pull/15171 and https://github.com/ARMmbed/mbed-os/pull/15232, support of the ST Nucleo H753ZI board should be included in mbed OS 6.16. However, a fresh copy of mbed Studio 1.4.5 that uses mbed OS 6.17 (according to the release notes) does not support this board. It is listed under "Other targets" (not the supported ones), and compiling code stops with the error message 
`argument -m/--mcu: NUCLEO_H753ZI is not a supported MCU. Supported MCUs are: [long list]`
Am I doing something wrong, or was support dropped again in 6.17?

#### Target(s) affected by this defect ?
ST Nucleo H753ZI

#### Toolchain(s) (name and version) displaying this defect ?
Standard (ARM compiler 6.16)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed Studio 1.4.5

#### How is this defect reproduced ? 

Change target, type in "753", Nucleo H753ZI will appear in the non-compatible section below. Compiling code will yield the error shown above.

[image]

JohnK1987, 2023-08-04:
Hello,

Support of Nucleo-H753ZI was never done. The first oen seems to be canceled before end and the second one add core files for the target as a MCU (STM32H753ZI) but not complete dev board.

How you can see [here](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H753xI) the folder for TARGET_NUCLEO_H753ZI is missing.

If these CORE files are correct and the rest is done then add the Nucleo-H753ZI iss not so hard.

BR, Jan

jeromecoutant, 2023-08-07:
Hi
Yes, Nucleo-H753ZI support was not merged. You need to rebase #15171 on top of master :-)
Thx

0xc0170, 2023-08-18:
I'll close this as resolved. Not supported.

closed 2023-08-18 by 0xc0170

supercrazysam, 2023-10-12:
Any plan or roadmap to get Nucleo-H753ZI  merged to mbed-os?  This would be very helpful.....

jeromecoutant, 2023-10-13:
Hi
As said, you need to rebase https://github.com/ARMmbed/mbed-os/pull/15171 on top of master, and propose a new PR.
Thx