ARMmbed/mbed-os #15444: Support for NUCLEO-H563ZI

sstaub, 2023-08-12:
Can you add support for the NUCLEO-H563ZI board?

### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2023-08-12:
@sstaub thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2023-08-16:
Hi
Sorry, STM32H5 support is not planned in ST side.

closed 2023-08-16 by 0xc0170