ARMmbed/mbed-os #15449: STM32F4 HAL is out of date

Maxoos, 2023-08-23:
### Description of defect

It seems that the HAL version included is out of date and newer boards are not full supported. I have a STM32F469I with an LCD and this LCD only seems to work with the latest cube hal drivers - [V1.8.1](https://github.com/STMicroelectronics/STM32CubeF4/commit/52757b5e33259a088509a777a9e3a5b971194c7d)

Current version included with mbed is 1.7.12 from two years ago.

#### Target(s) affected by this defect ?

STM32F4

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.17.0

mbedmain, 2023-08-23:
@Maxoos thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2023-08-24:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2023-08-24:
Hi
This is not planned in ST side,
but proposition via pull requests are welcomed!
Thx

arthurafarias, 2023-09-23:
Hi

I've upgraded the STM32Cube Firmware to the latest version 1.27.1.

Take a look at my pull request.

I did the same to F7 targets, I've update them too to the latest version.

I don't have any F4 board.

If you want to send me any I can test them if the update works for them.

Currently ST marked the current support for CAN and ETHERNET as legacy in these versions.

Maybe we need a new implementation of the drivers.

I will try to port in the next weekend the mbed-os to the new Ethernet and CAN hal.

Best regards.