ARMmbed/mbed-os #15450: Missing CDC_SEND_BREAK handling

yp05327, 2023-08-28:
### Description of defect

My PC side application needs to send a break signal to the board, but mbed-os will not handle it, and return failure.
I did some changes to the source for just making it return success, but I think it will be wonderful if this can be supported officially.

Related codes:
https://github.com/ARMmbed/mbed-os/blob/f0d909094097c6b6e201c6e7c87bcd3b64209f05/drivers/usb/source/USBCDC.cpp#L29-L31
↑ add the following codes.
``` c++
#define CDC_SEND_BREAK 0x23
```
https://github.com/ARMmbed/mbed-os/blob/f0d909094097c6b6e201c6e7c87bcd3b64209f05/drivers/usb/source/USBCDC.cpp#L218-L243
↑ add a new case in this `switch`
``` c++
case CDC_SEND_BREAK:
    /* Do something here */
    result = Success; 
    break; 
```

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2023-08-28:
@yp05327 thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

yp05327, 2023-10-10:
Updated the issue header with the missing information as much as possible.