ARMmbed/mbed-os #15452: STM32WL Deep Sleep STOP1 over consumption

hallard, 2023-09-05:
### Description of defect

We have a custom board with RAK3172 (STM32WL). We use a lot of different boards with this chip for low power and we never had issue with mbed going to consumption of approx 1.5uA un sleep mode (STOP2 default mbed for this target). We changed to go to STOP1 because we need LPTIM2 and the consumption is not the one expected in the datasheet and here

[image]

#### Target(s) affected by this defect ?

STM32WL

#### What version of Mbed-os are you using (tag or sha) ?

main and latest release, both

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed ide

[image]

#### How is this defect reproduced ? 

Change STOP2 to STOP1 in [sleep.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/sleep.c#L199-L201)

```cpp
    HAL_PWREx_EnterSTOP2Mode(PWR_STOPENTRY_WFI);
```
by 

```cpp
    HAL_PWREx_EnterSTOP1Mode(PWR_STOPENTRY_WFI);
```

in STOP2 our board is 1.5uA (expected) and in STOP1 it's 24uA instead of 7uA expected 

So I decided to give a try with ST official sample code for STM32WL [PWR_STOP2_RTC](https://github.com/STMicroelectronics/STM32CubeWL/blob/main/Projects/NUCLEO-WL55JC/Examples/PWR/PWR_STOP2_RTC/Src/main.c) doing same changes, ie change  `HAL_PWREx_EnterSTOP2Mode` to `HAL_PWREx_EnterSTOP1Mode` and then results are the one expected 

STOP2 is 1.5uA (don't change) and in STOP1 it's 7uA. I also activated LPTIM2 on ST Sample code to be sure, same results (7uA)

So I'm asking experts @jeromecoutant  @0xc0170 if we don't miss something for STOP1 mode (but as STOP2 is more 'cutting' things than STOP1 I'm curious how can it be possible.

We tried on severals boards with same issue.

Thanks

mbedmain, 2023-09-05:
@hallard thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

hallard, 2023-11-17:
This one is resolved it was related to a diode leak current / capacitor on specific boards, 7µA was good after few minutes so we put 10K in // of the capacitor (only powered when reading sensors) now STOP1 directly goes to 7µA

closed 2023-11-17 by hallard