ARMmbed/mbed-os #15460: TARGET_STM: masking logic for CAN rx interrupt is wrong

pavels, 2023-10-26:
### Description of defect

#### Target(s) affected by this defect ?

STM32

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

When there are multiple interrupts registered for CAN device, `rx` interrupt will not fire.

mbedmain, 2023-10-26:
@pavels thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

pavels, 2023-10-26:
And here is PR to fix the issue https://github.com/ARMmbed/mbed-os/pull/15461

closed 2023-11-23 by 0xc0170