ARMmbed/mbed-os #15476: Mbed CLI 2 USB no such file or directory

iccentt, 2023-12-15:
### Description of defect
I tried to compile USB application on KL25Z and K64F without success.
The error output is:
`main.cpp:2:10: fatal error: USBKeyboard.h: No such file or directory
    2 | #include "USBKeyboard.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!
`

#### Target(s) affected by this defect ?
KL25Z, K64F

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

6.17

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 2

#### How is this defect reproduced ? 
`
#include "mbed.h"
#include "USBKeyboard.h"

USBKeyboard key;

#define WAIT_TIME_MS 500 
DigitalOut led1(LED1);

int main()
{
    while (true)
    {
        led1 = !led1;
        thread_sleep_for(WAIT_TIME_MS);
    }
}
`

0xc0170, 2023-12-15:
Hi, 

how did you create the example ? I wonder if there is usb component being added. In CMake, I would link to `mbed-usb` component.

This answer might help you : https://forums.mbed.com/t/mbed-os-6-14-include-usbserial/14623

cyliangtw, 2023-12-21:
For CLI 2, the attached CMakeList.txt could build USBSerial or USBKeyboard successfully.

[CMakeLists.txt](https://github.com/ARMmbed/mbed-os/files/13734307/CMakeLists.txt)

closed 2023-12-22 by 0xc0170