ARMmbed/mbed-os #15480: STM32-gen_peripheralPins.py chrashing after generating pinmaps

lefebvresam, 2023-12-30:
### Description of defect

When generating pinmap files the script is crashing:

```
mbed-os$ python targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py -m "STM32U575RGTx.xml"

Script version 1.20.5

Checking STM32_open_pin_data repo...
        Already up to date.

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.90

 * Output directory: /home/samlef/git/hmipanel/mbed-os/TARGET_CUSTOM/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xG/TARGET_STM32U575RGT
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data/mcu/STM32U575RGTx.xml'
 * GPIO file: STM32_open_pin_data/mcu/IP/GPIO-STM32U5x_gpio_v1_0_Modes.xml
Traceback (most recent call last):
  File "targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py", line 1933, in <module>
    parse_pins()
  File "targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py", line 1399, in parse_pins
    if "Variant" in s.attributes:
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 522, in __getitem__
    return self._attrs[attname_or_tuple]
KeyError: 0
```

#### Target(s) affected by this defect ?

Custom target

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

Mbed 7.59.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

CLI2

#### How is this defect reproduced ? 

Execute command above

lefebvresam, 2024-01-18:
Handled at https://github.com/mbed-ce/mbed-os/issues/212

closed 2024-01-18 by lefebvresam

jeromecoutant, 2024-01-19:
Hi
I need more info to reproduce:

````
$ python targets/TARGET_STM/tools/STM32_gen_PeripheralPins.py -m "STM32U575RGTx.xml"

Script version 1.20.5

Checking STM32_open_pin_data repo...
        Already up to date.

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.90

 * Output directory: xxx\TARGET_CUSTOM\TARGET_STM\TARGET_STM32U5\TARGET_STM32U575xG\TARGET_STM32U575RGT
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data\mcu\STM32U575RGTx.xml'
 * GPIO file: STM32_open_pin_data\mcu\IP\GPIO-STM32U5x_gpio_v1_0_Modes.xml
 * I/O pins found: 120 connected: 1
````

lefebvresam, 2024-01-19:
Was probably due to Python2, see https://github.com/mbed-ce/mbed-os/issues/212