ARMmbed/mbed-os #15482: BLE start routine get freeze when the event_queue dispatching inside mbed-os-ble-utils moved to a thread

16L-YT, 2024-01-04:
### Description of defect
I'm using mbed-os-ble-utils for our BLE-enabled device. In this library the `BLEProcess` class has the `start` API to strart advertising. I tried to fork the library and make the `_event_queue.dispatch_forever();` ([line](https://github.com/ARMmbed/mbed-os-ble-utils/blob/0da6cf31bd6688e4a3ecdf98b356e4767f3248c4/ble_process.h#L90C9-L90C41)) inside a thread to avoid blocking the code I will execute at startup after starting the BLE advertisements .

```
        thread.start([this] {_event_queue.dispatch_forever();});
```

However the BLE stuck at  

> [DBG ][BLE ]: Cordio HCI reset started

I even tried to increase the priority of that thread to the highest `osPriorityRealtime7` with no effect.  

Note: 
No extra debugging messages, even if the BLE logs are enabled: 

```
            "cordio.trace-hci-packets": false,
            "cordio.trace-cordio-wsf-traces": true,
            "ble.trace-human-readable-enums": true,
```

Is there a clear reason why ? 

#### Target(s) affected by this defect ?
AMA3B1KK-KBR-B0

#### Toolchain(s) (name and version) displaying this defect ?
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 7.59.0

#### How is this defect reproduced ? 
Following the mentioned steps in the defect description.

16L-YT, 2024-01-05:
Changing `_event_queue.dispatch_forever()` to `_event_queue.dispatch()` solved the problem. Then you can call `thread.start([this] {_event_queue->dispatch_forever();});`

closed 2024-01-05 by 16L-YT

yahyatawil, 2024-02-15:
Just for future reference, my suggestion was not accurate and not even functional in my last comment. It seems the Bluetooth APIs are not thread-safe according to #14558. I don't know how to make the BLE nonblocking!