ARMmbed/mbed-os #15486: PD_14 is not labeled as a SPI CS pin for Nucleo-F413ZH

Archerist, 2024-01-16:
### Description of defect

[Nucleo 144 manual](https://www.st.com/resource/en/user_manual/um1974-stm32-nucleo144-boards-mb1137-stmicroelectronics.pdf) specifies that PD14 (D10) pin can be used as a SPI CS pin, however using PD_14 in SPI api as `ssel` produces this error:
```
++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: pinmap not found for peripheral
Location: 0x800684F
Error Value: 0x3E
Current Thread: application_unnamed_thread Id: 0x20002A54 Entry: 0x800807D StackSize: 0x1000 StackMem: 0x20005F28 SP: 0x20006E34 
For more info, visit: https://mbed.com/s/error?error=0x80010130&tgt=NUCLEO_F413ZH
```

[Board Page](https://os.mbed.com/platforms/ST-Nucleo-F413ZH/#board-pinout) is also missing the label.

#### Target(s) affected by this defect ?
Nucleo-F413ZH

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.17.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
I use PlatformIO
```
Platform ststm32 @ 17.2.0 (required: ststm32)
├── framework-mbed @ 6.61700.231105 (required: platformio/framework-mbed @ ~6.61700.0)
├── tool-dfuutil @ 1.11.0 (required: platformio/tool-dfuutil @ ~1.11.0)
├── tool-dfuutil-arduino @ 1.11.0 (required: platformio/tool-dfuutil-arduino @ ~1.11.0)
├── tool-openocd @ 3.1200.0 (required: platformio/tool-openocd @ ~3.1200.0)
├── tool-stm32duino @ 1.0.1 (required: platformio/tool-stm32duino @ ~1.0.1)
└── toolchain-gccarmnoneeabi @ 1.90201.191206 (required: platformio/toolchain-gccarmnoneeabi @ ~1.90201.0)
```

#### How is this defect reproduced ? 
Use PD_14 in SPI API's `ssel` parameter

jeromecoutant, 2024-01-19:
Hi
Yes, there is no pin with a HW SPI CS pin in D10:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F413xH/TARGET_NUCLEO_F413ZH/PeripheralPins.c#L358-L373
You need to use D10 as a SW SPI CS

Jerome

Archerist, 2024-01-25:
I don't understand, Nucleo 144 manual literally labels it as a SPI CS pin
[image]

jeromecoutant, 2024-01-25:
Yes, we can say it is a bug in this User Manual...
Reference is MCU data sheet: https://www.st.com/resource/en/datasheet/stm32f413zh.pdf
[image]

Check https://github.com/ARMmbed/mbed-os/blob/master/drivers/include/drivers/SPI.h#L61

Archerist, 2024-01-25:
Got it, thanks for your time

closed 2024-01-25 by Archerist