ARMmbed/mbed-os #15488: PC_13 is unlisted

G4GNX, 2024-01-18:
Page <https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/TARGET_NUCLEO_F767ZI/PeripheralPins.c>
does not list PC_13 or "User_Button" which is attached to PC13

[Board Page](https://os.mbed.com/platforms/ST-Nucleo-F767ZI/#board-pinout) does contain the label.

So does page <https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F767xI/TARGET_NUCLEO_F767ZI/PinNames.h>

Target(s) affected by this defect ?
Nucleo-F767ZI

Toolchain(s) (name and version) displaying this defect ?
None

mbedmain, 2024-01-18:
@G4GNX thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2024-01-19:
Hi
PeripheralPins.c file only reference pins for ADC, PWM, UART,..etc
PC13 is "only" a button (wake up pin)

closed 2024-01-28 by G4GNX