ARMmbed/mbed-os #15490: Update hidapi used for non-Linux platforms

makinbacon21, 2024-02-13:
### Description of defect

`cython-hidapi` v0.7.x is no longer maintained--a one-time fixup commit (https://github.com/trezor/cython-hidapi/commit/15dcc129e5c3de603f717d5f99b56693fbae91d6) fixes building, but before this the Mbed OS tools did not even compile on non-Linux platforms. I'm not sure exactly what relies on this version of the API (didn't have time to investigate), but this dep should be updated regardless. Please refer to https://github.com/trezor/cython-hidapi/pull/173#issuecomment-1942067223 for more information.

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

Latest tag and master branch are affected

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Latest mbed-cli, tested on Python 3.10 and 3.11 (much more broken on 12).

#### How is this defect reproduced ? 

Try to run `pip install -r requirements.txt` on a non-Linux platform

ApplicationNinja, 2024-02-24:
Need this fixed...

0xc0170, 2024-02-25:
There are usb tests that requiere this hidapi package https://github.com/ARMmbed/mbed-os/commit/852390fcdcfd030325301fc4e74e18fa5e71ec6e

What versions should we update to?