ARMmbed/mbed-os #15495: Flashing Controller from Docker Container

SebastianZug, 2024-03-06:
### Description of defect

I would like to flash a microcontroller with mbed code from a Docker container.  The solution describe in [mbed documation](ttps://os.mbed.com/docs/mbed-os/v6.16/build-tools/docker.html) mentioned by [@saheerb](https://github.com/ARMmbed/mbed-tools/issues/308#issuecomment-913630606) does not work any more. When I run the container by

```
docker run -it --privileged -v /dev/disk/by-id:/dev/disk/by-id -v /dev/serial/by-id:/dev/serial/by-id -v /run/udev:/run/udev:ro ghcr.io/armmbed/mbed-os-env:master-latest
```

I receive an error  

```
> mount /dev/sda /mnt  
> mbedls

root@1804d0877b61:~# mbedls
WARNING:mbedls.platform_database:Error loading database /root/.local/share/mbedls/platforms.json: Platform Database is out of date; Recreating
Traceback (most recent call last):
  File "/usr/local/bin/mbedls", line 8, in <module>
    sys.exit(mbedls_main())
  File "/usr/local/lib/python3.8/dist-packages/mbed_lstools/main.py", line 196, in mbedls_main
    ret_code = args.command(mbeds, args)
  File "/usr/local/lib/python3.8/dist-packages/mbed_lstools/main.py", line 62, in print_table
    return print_mbeds(mbeds, args, False)
  File "/usr/local/lib/python3.8/dist-packages/mbed_lstools/main.py", line 47, in print_mbeds
    devices = mbeds.list_mbeds(unique_names=True, read_details_txt=True)
  File "/usr/local/lib/python3.8/dist-packages/mbed_os_tools/detect/lstools_base.py", line 135, in list_mbeds
    candidates = list(self.find_candidates())
  File "/usr/local/lib/python3.8/dist-packages/mbed_os_tools/detect/linux.py", line 51, in find_candidates
    disk_ids = self._dev_by_id("disk")
  File "/usr/local/lib/python3.8/dist-packages/mbed_os_tools/detect/linux.py", line 76, in _dev_by_id
    to_ret = dict(
  File "/usr/local/lib/python3.8/dist-packages/mbed_os_tools/detect/linux.py", line 117, in _hex_ids
    yield match.group("usbid"), _readlink(dl)
  File "/usr/local/lib/python3.8/dist-packages/mbed_os_tools/detect/linux.py", line 31, in _readlink
    content = os.readlink(link)
OSError: [Errno 22] Invalid argument: '/dev/disk/by-id/usb-MBED_microcontroller_066BFF303555483043215322-0'
```

For `mbed-tools detect` we have an additional compatibility problem with python packages

```
root@1804d0877b61:~# mbed-tools detect
Traceback (most recent call last):
  File "/usr/local/bin/mbed-tools", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/cli/main.py", line 38, in invoke
    super().invoke(context)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/cli/list_connected_devices.py", line 29, in list_connected_devices
    connected_devices = get_connected_devices()
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/devices/devices.py", line 24, in get_connected_devices
    for candidate_device in detect_candidate_devices():
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/devices/_internal/detect_candidate_devices.py", line 16, in detect_candidate_devices
    detector = _get_detector_for_current_os()
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/devices/_internal/detect_candidate_devices.py", line 27, in _get_detector_for_current_os
    from mbed_tools.devices._internal.linux.device_detector import LinuxDeviceDetector
  File "/usr/local/lib/python3.8/dist-packages/mbed_tools/devices/_internal/linux/device_detector.py", line 11, in <module>
    import pyudev
  File "/usr/local/lib/python3.8/dist-packages/pyudev/__init__.py", line 48, in <module>
    from pyudev.core import Context, Enumerator
  File "/usr/local/lib/python3.8/dist-packages/pyudev/core.py", line 39, in <module>
    from pyudev.device import Devices
  File "/usr/local/lib/python3.8/dist-packages/pyudev/device/__init__.py", line 26, in <module>
    from ._device import Attributes, Device, Devices, Tags
  File "/usr/local/lib/python3.8/dist-packages/pyudev/device/_device.py", line 37, in <module>
    from six.moves import collections_abc
ImportError: cannot import name 'collections_abc' from 'six.moves' (unknown location)
```

#### Target(s) affected by this defect ?

I am using an [STM32 B-L475E-IOT0A1](https://www.st.com/en/evaluation-tools/b-l475e-iot01a.html) board.

```
[717662.825303] usb 3-3.1: new full-speed USB device number 9 using xhci_hcd
[717662.975484] usb 3-3.1: New USB device found, idVendor=0483, idProduct=374b, bcdDevice= 1.00
[717662.975487] usb 3-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[717662.975489] usb 3-3.1: Product: STM32 STLink
[717662.975490] usb 3-3.1: Manufacturer: STMicroelectronics
[717662.975490] usb 3-3.1: SerialNumber: 066BFF303555483043215322
[717663.055283] usb-storage 3-3.1:1.1: USB Mass Storage device detected
[717663.056208] scsi host0: usb-storage 3-3.1:1.1
[717663.056718] cdc_acm 3-3.1:1.2: ttyACM0: USB ACM device
[717664.057979] scsi 0:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
[717664.058169] sd 0:0:0:0: Attached scsi generic sg0 type 0
[717664.058654] sd 0:0:0:0: [sda] 6216 512-byte logical blocks: (3.18 MB/3.04 MiB)
[717664.058935] sd 0:0:0:0: [sda] Write Protect is off
[717664.058937] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[717664.059274] sd 0:0:0:0: [sda] No Caching mode page found
[717664.059277] sd 0:0:0:0: [sda] Assuming drive cache: write through
[717664.077104] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

#### Toolchain(s) (name and version) displaying this defect ?

The image is 2 years old and available at [mbed-os-env:master-2022.05.21t04.23.55](https://github.com/armmbed/mbed-os/pkgs/container/mbed-os-env).

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.15.0 is mentioned in the container description.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

I tested both mbed-cli and mbed-tools.

mbedmain, 2024-03-06:
@SebastianZug thank you for raising this issue.Please take a look at the following comments:

How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

SebastianZug, 2024-03-06:
Done :-)