ARMmbed/mbed-os #15505: LoRaWAN no OTAA Uplink on AS923 region STM32WL

shiwzhi, 2024-04-10:
### Description of defect

Not be able to use AS923 region, no OTAA received on gateway on frequency 923.2Mhz and 923.4Mhz

#### Target(s) affected by this defect ?
STM32WLE5 RAK3172

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

17dc3dc2e6e2817a8bd3df62f38583319f0e4fed

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.4.5

#### How is this defect reproduced ? 

```
{
    "config": {
        "main_stack_size": {
            "value": 16000
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 9600,
            "platform.minimal-printf-enable-floating-point": 1,
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": false,
            "lora.duty-cycle-on-join": true,
            "lora.phy": "AS923",
            "lora.phy-as923-sub-region": "AS1",
            "lora.downlink-preamble-length": 8
        },
        "RAK3172": {
            "stm32wl-lora-driver.crystal_select": 0,
            "target.stdio_uart_tx": "PA_2_ALT0",
            "target.stdio_uart_rx": "PA_3_ALT0",
            "platform.cpu-stats-enabled": 1
        }
    },
    "macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_lora_config.h\""
    ]
}
```
AU915 and US915 works, so should not be the problem with the hardware and gateway, but not AS923, and no problem with RAK's AT firmware

mbedmain, 2024-04-10:
@shiwzhi thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

cyliangtw, 2024-06-03:
I verified the AS923 OK on NUMAKER_IOT_M252 + SX1276 based on OTAA mode of TTN server.
Just for reference:
```
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
	    "platform.minimal-printf-enable-floating-point" : true,
            "mbed-trace.enable": true,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "lora.over-the-air-activation": true,
            "lora.duty-cycle-on": true,
            "lora.phy": "AS923",
	    "lora.max-sys-rx-error": 16,
            "lora.fsb-mask": "{0xFFFF}",
            "lora.device-eui": "{ your-dev-eui }",
            "lora.application-eui": "{ your-ap-eui }",
            "lora.application-key": "{ your-app-key }"
        },
```

shiwzhi, 2024-06-06:
> I verified the AS923 OK on NUMAKER_IOT_M252 + SX1276 based on OTAA mode of TTN server. Just for reference:
>
> ```
>     "target_overrides": {
>         "*": { [...]

Thank you, after I change the antenna gain to a smaller value, I was able to receive uplink on the gateway
```
"macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_lora_config.h\"",
        "LORAPHY_ANTENNA_GAIN=-30.0f"
    ]
```

closed 2024-06-06 by shiwzhi